lexus2k/ssd1306 #99: Pins

moo06, 2020-01-12:
Hi,
Is there any way to change the pin outputs for the OLED. I'm making a game console for a school project and need to use the one of the pins that the OLED uses. Some OLED libraries used pins 5 and 7 but they are also not as good as this one.

Thanks,
Merle

lexus2k, 2020-01-12:
Hi,
It depends on platform, you're using in your project?
Could you please send me IDE, you're using and micro-controller name?
https://lexus2k.github.io/ssd1306/structssd1306__platform__i2c_config__t.html : `ssd1306_platform_i2cConfig_t` allows to choose pin numbers

moo06, 2020-01-15:
Hi,
I got it to work, thank you so much. I also have another question which ill ask on a separate issue.

Thanks

closed 2020-01-15 by moo06