lexus2k/ssd1306 #100: Numbers

moo06, 2020-01-15:
Hi,
Is there any way I could print a variable onto the OLED?

Thanks

lexus2k, 2020-01-16:
Hi,

what type of variable you're trying to print?

moo06, 2020-01-16:
Hi,
It is an analogRead() because I'm debugging the button values so I can make a tech demo. Since I'm using an ATTINY85, and I'm using practically all the pins, I can only try and show it on the screen.

Thanks

lexus2k, 2020-01-16:
Hi here is an example for printing numbers:

```.cpp
#include "ssd1306.h"
#include <stdio.h>

void printNumber(int x, int y, int number)
{
    char intStr[12];
    snprintf(intStr, sizeof(intStr), "%i", number );
    ssd1306_printFixed (x, y, intStr, STYLE_NORMAL);
}
```

lexus2k, 2020-01-25:
Did the example above help you?

moo06, 2020-01-25:
Hi, yes thank you.

closed 2020-01-25 by moo06