lexus2k/ssd1306 #103: low power mode

philippedc, 2020-02-09:
Hi Aleksei, I'm wondering if there is a better way than _`ssd1306_displayOff()`_ to put the oled display in low power mode when the Attiny85 is in sleep mode ?
Strangely I have 3 identical circuits based around an Attiny85, a BME280 and an oled SSD1306. In sleep mode I have one with 0.22mA, another with 0.25mA and one with 0.011mA only.
I cannot understand why I get 20 time less power for this last one, except the oled itself.

lexus2k, 2020-02-10:
Hi Philipp,

`0xAE` command is the only way to put ssd1306 display to sleep mode. According to datasheet in sleep mode ssd1306 controller power consumption should be less than 10uA for Vdd (logic power), and less than 10uA for Vcc (panel power) without panel attached. That means, that values from datasheet depend on display panel manufacturer. But anyway OLED display panel should not consume itself much power, if no pixels are ON. For display ON modes ssd1306 datasheet says that logic consumption is upto 150uA, and panel consumption is upto 780uA. Maybe, on 2 circuits the display actually doesn't go to sleep more for some reason?

Another thing: how did you implement pull up for i2c lines in your schematics?

philippedc, 2020-02-10:
Many thanks for your quick reply !
Here is the code:

`/*
Station météorologique pour ATtiny85 à économie d'énergie

pour la partie OLED écran I2C :
https://github.com/lexus2k/ssd1306
pour la partie sonde BME280 :
https://github.com/conurb/low_energy_sensor/blob/master/low_energy_sensor.ino
et pour la partie SLEEP : 
https://github.com/blevien/attiny85-sleep/blob/master/attiny85-sleep.ino
https://gist.github.com/JChristensen/5616922
ATTENTION attachInterrupt(0, poussoir, LOW) gère par défaut les interruption en INT0, 
cad en PB2 déjà utilisé pour le SCL
D'où l'astuce pour changer la commande de l'interruption :
https://www.insidegadgets.com/2011/02/27/how-to-use-the-pin-change-interrupt-on-attiny85/

                        +-------+
                       1|*      |8  VCC
                  PB3  2|       |7  PB2 = SCL (BME280 + OLED)
                  PB4  3|       |6  PB1 => LED
                  GND  4|       |5  PB0 = SDA (BME280 + OLED)
                        +-------+
 connection I2C :
 SDA en PB0 = pin5
 SCL en PB2 = pin7
  A3 -> PB3 = pin2 => détecteur d'humidité
 LED en PB1 = pin6 => led d'avertissement de sécheresse
 SWI en PB4 = pin3 => push-button to wake-up
*/

#include <avr/sleep.h>
#include <avr/interrupt.h>
#include <avr/power.h>    
#include <EEPROM.h>
#include "ssd1306.h"            // https://github.com/lexus2k/ssd1306
#include "bme280_tiny_i2c.h"    // https://github.com/conurb/low_energy_sensor

#ifndef cbi
#define cbi(sfr, bit) (_SFR_BYTE(sfr) &= ~_BV(bit))
#endif
#ifndef sbi
#define sbi(sfr, bit) (_SFR_BYTE(sfr) |= _BV(bit))
#endif

int v;
int memo[]={0,0,0};
char chaine[20];
bool r1=false;
volatile bool r2=false;
byte j;

BME280 bme;

void setup() {
  pinMode(1, OUTPUT);          // LED
  pinMode(3, INPUT_PULLUP);    // pin2 ground humidity level sensor
  pinMode(4, INPUT_PULLUP);    // pin3 to push-button

  j = EEPROM.read(3);

  //bme280_init(&bme);
  
  sbi(GIMSK,PCIE);    // Turn on Pin Change interrupt
  sbi(PCMSK,PCINT4);  // Which pins are affected by the interrupt

// the watch dog sets the interval delay :
// 0=16ms, 1=32ms,2=64ms, 3=128ms, 4=250ms, 5=500ms, 6=1s, 7=2s, 8=4s, 9=8s
  setup_watchdog(9);
[... 140 lines cut ...]
  EEPROM.write(3, j);         // save the counter 
  power_all_disable();        // power off ADC, Timer 0 and 1, serial interface
  sleep_enable();
  sleep_cpu();                // System actually sleeps here
  sleep_disable();            // System continues execution here when watchdog timed out 
  power_all_enable();         // power everything back on
}

void setup_watchdog( byte tempo ) {
  byte val;
  val = tempo & 7;
  if (tempo > 7) val |= (1<<5);
  val |= (1<<WDCE);
  MCUSR &= ~(1<<WDRF);                  // start timed sequence
  WDTCR |= (1<<WDCE) | (1<<WDE);        // set new watchdog timeout value
  WDTCR = val;
  WDTCR |= _BV(WDIE);
}
  
// Watchdog Interrupt Service / is executed when watchdog timed out
ISR(WDT_vect) { r1 = true; r2 = false; }        // set global flag  

// External Interrupt Service / is executed when push-button is pressed
ISR(PCINT0_vect) { r2 = true;}      // set global flag  
`

philippedc, 2020-02-10:
Many Thanks for your quick reply !
Here is the code:
`/*
Station météorologique pour ATtiny85 à économie d'énergie

pour la partie OLED écran I2C : https://github.com/lexus2k/ssd1306
pour la partie sonde BME280 :
https://github.com/conurb/low_energy_sensor/blob/master/low_energy_sensor.ino
et pour la partie SLEEP : 
https://github.com/blevien/attiny85-sleep/blob/master/attiny85-sleep.ino
https://gist.github.com/JChristensen/5616922
ATTENTION attachInterrupt(0, poussoir, LOW) gère par défaut les interruption en INT0, 
cad en PB2 déjà utilisé pour le SCL
D'où l'astuce pour changer la commande de l'interruption :
https://www.insidegadgets.com/2011/02/27/how-to-use-the-pin-change-interrupt-on-attiny85/

                        +-------+
                       1|*      |8  VCC
                  PB3  2|       |7  PB2 = SCL (BME280 + OLED)
                  PB4  3|       |6  PB1 => LED
                  GND  4|       |5  PB0 = SDA (BME280 + OLED)
                        +-------+
 connection I2C :
 SDA en PB0 = pin5
 SCL en PB2 = pin7
  A3 -> PB3 = pin2 => détecteur d'humidité
 LED en PB1 = pin6 => led d'avertissement de sécheresse
 SWI en PB4 = pin3 => push-button to wake-up
*/

#include <avr/sleep.h>
#include <avr/interrupt.h>
#include <avr/power.h>    
#include <EEPROM.h>
#include "ssd1306.h"            // https://github.com/lexus2k/ssd1306
#include "bme280_tiny_i2c.h"    // https://github.com/conurb/low_energy_sensor

#ifndef cbi
#define cbi(sfr, bit) (_SFR_BYTE(sfr) &= ~_BV(bit))
#endif
#ifndef sbi
#define sbi(sfr, bit) (_SFR_BYTE(sfr) |= _BV(bit))
#endif

int v;
int memo[]={0,0,0};
char chaine[20];
bool r1=false;
volatile bool r2=false;
byte j;

BME280 bme;

void setup() {
  pinMode(1, OUTPUT);          // LED
  pinMode(3, INPUT_PULLUP);    // pin2 ground humidity level sensor
  pinMode(4, INPUT_PULLUP);    // pin3 to push-button

  j = EEPROM.read(3);

  //bme280_init(&bme);
  
  sbi(GIMSK,PCIE);    // Turn on Pin Change interrupt
  sbi(PCMSK,PCINT4);  // Which pins are affected by the interrupt

// the watch dog sets the interval delay :
// 0=16ms, 1=32ms,2=64ms, 3=128ms, 4=250ms, 5=500ms, 6=1s, 7=2s, 8=4s, 9=8s
  setup_watchdog(9);
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);  // sleep mode available
    
[... 138 lines cut ...]
  EEPROM.write(3, j);         // save the counter 
  power_all_disable();        // power off ADC, Timer 0 and 1, serial interface
  sleep_enable();
  sleep_cpu();                // System actually sleeps here
  sleep_disable();            // System continues execution here when watchdog timed out 
  power_all_enable();         // power everything back on
}

void setup_watchdog( byte tempo ) {
  byte val;
  val = tempo & 7;
  if (tempo > 7) val |= (1<<5);
  val |= (1<<WDCE);
  MCUSR &= ~(1<<WDRF);                  // start timed sequence
  WDTCR |= (1<<WDCE) | (1<<WDE);        // set new watchdog timeout value
  WDTCR = val;
  WDTCR |= _BV(WDIE);
}
  
// Watchdog Interrupt Service / is executed when watchdog timed out
ISR(WDT_vect) { r1 = true; r2 = false; }        // set global flag  

// External Interrupt Service / is executed when push-button is pressed
ISR(PCINT0_vect) { r2 = true;}      // set global flag  
`

lexus2k, 2020-02-10:
Hello,

why don't you call `sd1306_clearScreen( );` and `ssd1306_displayOff();` before entering sleep mode in your example?
The `system_sleep()` code looks OK: you set sleep mode `SLEEP_MODE_PWR_DOWN`, then disable ADC, sleep_enable(), and perform sleep_cpu().

And could you please, clarify? If you say that you have 3 devices, and 2 of them consume too much power, then maybe the root cause is hardware issue?

philippedc, 2020-02-10:
I call `ssd1306_clearScreen( );` and `ssd1306_displayOff();` once I do not need to display anything, so in the same loop() sequence than initialization with `ssd1306_128x64_i2c_init();` etc...
Hoever you're right, I will try by repeating the displayOff instruction in the sleep function.
Thanks

philippedc, 2020-02-10:
So I made tests : when I insert `ssd1306_clearScreen( );` and `ssd1306_displayOff();` in the sleep function, I get a sleep current of 1mA. After I push the wakeup button, the next sleeping mode gives 0,24mA.
So it can be said that these 2 instructions without a ssd1306 initialisation will put the ssd1306 in a mode that it will have a current about 4 times the "normal" sleeping current.
So it is worst than expected.

lexus2k, 2020-02-11:
Hello,

I've performed tests for ssd1306 display (i2c, 128x64) and `ssd1306_displayOff()`/`ssd1306_displayOn()` functions. And here are my results:
1. When display is on and fully enabled, its power consumption is 10.79mA. So, that is expected.
2. When I send ssd1306_displayOff() command it's power consumption is 5uA (this is also expected).

So, I would recommend you to check schematics, and pcb assembly.

philippedc, 2020-02-11:
You are right, I have done the same tests, with similar results. 
I have ordered new BME280 to be able to test them separately.
As the circuit is only a BME280, a SSD1306 and one LED, the circuit is quickly checked... 
Many thanks for your help.

lexus2k, 2020-02-13:
Let me know, when you find the root cause. Thank you

lexus2k, 2020-03-25:
Any updates on low power mode?

philippedc, 2020-03-25:
Hi @lexus2k I confirm the over power consumption comes from the BME280.
Thanks

lexus2k, 2020-04-05:
Thank you very much

closed 2020-04-05 by lexus2k