lexus2k/ssd1306 #105: How to display double or floats

sounox, 2020-05-26:
Hello, first of all thank you for making this librairy, for a project i need to display 
floating point number which incremented or decremented with a rotary encoder, i saw how to do it with int but not with floating point numbers!
Thank you !

lexus2k, 2020-05-26:
Hello,

For now the library doesn't has built-in function to display float numbers, but here is an example how to to that:
```.cpp
#include "ssd1306.h"
#include <stdio.h>

void printFloatNumber(int x, int y, float number)
{
    char intStr[16];
    snprintf(intStr, sizeof(intStr), "%f", number );
    ssd1306_printFixed (x, y, intStr, STYLE_NORMAL);
}
```

sounox, 2020-05-26:
Thanks for the fast answer! sadly i can't get it to work, it just freezes the atmega and display a "?".

ImpulseAdventure, 2020-05-26:
By default, Arduino does not include full support for floating point in printf(). Try using dtostrf() instead.

sounox, 2020-05-26:
The dtostrf slows waaay too much my code which makes it "skip" the inputs from the encoder :/

ImpulseAdventure, 2020-05-26:
If you are only incrementing a floating point number with an encoder input then I would definitely suggest looking into the option of eliminating your floats altogether and replacing them with a fixed-point notation using integers.

In other words, if you need at most 2 decimal places, then your internal counter uses “1562” to represent “15.62” and you simply use division/100 when converting back to a “floating point” string, perhaps something like the following:

sprintf(str,”%03d.%02d”, myVal/100, myVal%100)

Alternately, you can also use floats and perform a similar means of constructing the string by parts as above.

sounox, 2020-05-26:
Thank you all for your help ! I kind of optimised my code and used dtostrf, have a nice day !

closed 2020-05-26 by sounox