lexus2k/ssd1306 #106: Problem using the 'negativeMode' in a menu loop

leslieadams57, 2020-05-30:
**Describe the bug**
When using the library to display a 2 dimensional menu I use the 'negativeMode' to highlight the active menu entry. When scrolling through the option when reaching the end the counter is reset to the beginning. However instead of just highlighting the menu option the whole screen (except other menu entries which have been re-displayed using positiveMode) comes up negative.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to first menu level and scroll through to bottom and back to top.
2. Click on '....' n/a
3. Scroll down to '....' n/a
4. See error

**Expected behaviour**
When scrolling through the menu only the active item should be highlighted.

**Screenshots**
First time through is correct:
[image]
Second time is wrong:
[image]

**Please complete the following information:**
 - library version - 1.8.2
 - LCD display type - SSD1306
 - OS [e.g. linux, windows] - Windows 10
 - Platform [e.g. Atmega328p, esp32, etc.] - Atmega328p (5v 16Mhz Pro Mini)
 - IDE if using some - Arduino 1.8.12

**Additional context**
Copy of the code for the writeMenu() function:

int lvcnt = 0;               // menu level count
int itemcnt[]={0,0,0};       // current item within each level
int itemmax[]={3,2,3};       // max for array item  for each level note count starts at 0 e.g. 0-3 =4

const char *menuList[3][4]            {{"Settings","Run","Replay","Reset"},
                             {"Type","Calibrate","Return"," "},
                             {"SBUS","CPPM","Buddy Lead","Return"}};

const char *menuTitle[3]             {"Main Menu","Settings Menu","TX Type menu"};

void displayMenu(){                       		// display menu on OLED screen
  ssd1306_clearScreen();
  ssd1306_positiveMode();                             	// set default display mode
  ssd1306_printFixed(18, 0, "ThrustMeter", STYLE_BOLD);
  ssd1306_printFixed(20, 9, menuTitle[lvcnt], STYLE_BOLD);
  for (uint8_t f=0; f<(itemmax[lvcnt]+1); f++){         // loop around levels
    ssd1306_positiveMode();                             // set default display mode
    if (f == itemcnt[lvcnt]) {                          // check for the active item
      ssd1306_negativeMode();                             	// set active item negative display mode
    } 
    ssd1306_printFixed(20, (f*8)+24, menuList[lvcnt][f], STYLE_NORMAL);
  }
}

lexus2k, 2020-05-31:
Hello, there is no bug in the library. Just swap 2 lines in your code:

```.cpp
void displayMenu(){ // display menu on OLED screen
    ssd1306_positiveMode(); // set default display mode
    ssd1306_clearScreen();
    ssd1306_printFixed(18, 0, "ThrustMeter", STYLE_BOLD);
    ssd1306_printFixed(20, 9, menuTitle[lvcnt], STYLE_BOLD);
...
```

`ssd1306_positiveMode()` should be called before `ssd1306_clearScreen()`.

closed 2020-05-31 by lexus2k

leslieadams57, 2020-05-31:
Many thanks for that, I can confirm swapping the 2 lines did work. What I'm not sure of is why.
I thought setting positive mode only applied to any prints carried out, but from what you're saying it applies to the whole screen e.g. if neg is set and then you clearscreen the whole screen is set neg.
I am right in assuming the clearscreen function does not apply to attributes (e.g. neg or pos) you've set but in fact uses them to set the 'clear' mode?
Apart from this one minor problem I've been using your library for just a short while but I finding easy and relevant to use. Thanks very much.

lexus2k, 2020-06-02:
Positive and negative modes apply to all graphics functions for ssd1306 displays.
It is software implementation. Also, ssd1306 supports some hardware support for negative/positive, but that's different story. 
clearScreen function just fills GDRAM memory with 0x00 (positive) or 0xFF (negative) mode.
Thank you