lexus2k/ssd1306 #108: drawBitmap1 doesn't draw with Attiny85

mstratman, 2020-07-27:
**Describe the bug**
Using a NanoEngine1, `engine.canvas.drawBitmap1(...)` does not draw anything to the screen when using an AtTiny85. It works as expected with an Arduino Nano (atmega328p).

**To Reproduce**

https://gist.github.com/mstratman/52d323705083ee226dce1b6c766b9d89

**Expected/Actual behavior**

Expected, works correctly on the atmega Nano: bitmap of spaceship is drawn.

With attiny85 you only see the line resulting from the `drawLine()` (which was added to rule out hardware wiring or programming problems), but you do not see the bitmap 

**Please complete the following information:**
 - library version: **1.8.2**
 - LCD display type: **0.96 OLED, 128 x 64, I2C**
 - OS [e.g. linux, windows]: Mac
 - Platform [e.g. Atmega328p, esp32, etc.]: attiny85
 - IDE if using some: Arduino (avrdude)

lexus2k, 2020-07-27:
Thanks you for catching the bug. I confirm that for Attiny85 I observe the same issue.
Will come back to you with solution

lexus2k, 2020-07-28:
Please check the fix https://github.com/lexus2k/ssd1306/tree/1.8_dev

mstratman, 2020-07-28:
That seems to have worked, thank you!

closed 2020-08-04 by lexus2k