lexus2k/ssd1306 #109: Online Arduino Playground for SSD1306 display

Puneethkumarsn, 2020-08-08:
Hi There,

I am very happy to stumble upon your library for [SSD1306](https://wokwi.com/playground/ssd1306). We at wokwi.com have created an online Arduino playground and it can be based on your library as well. We will be able to create a PR shortly.

This will help the user of this library to visualize their code with the simulated OLED display element. This will also help them to gain more confidence in the library.

we will take care of maintaining the documentation on how to use the Arduino playground as well. Please leave a comment and we will be happy to raise a PR. How does it sound?
Arduino playground is open source and free for the users.

lexus2k, 2020-08-10:
Hello,

Thank you for your interest to the library. You did a great job. I'm very glad that you found it useful.
The sketch, you specified above uses Adafruit ssd1306. Do you use ssd1306 library for display emulation?

Puneethkumarsn, 2020-08-11:
Hi there,

sorry for not making it clear. The shared link had adafruit ssd1306. It is very easy to change the example code to work for our repo example. 

[To make it easier, I have attached a small video showing - How to run the example codes from your library in the Arduino playground
](https://drive.google.com/file/d/11V6Xu3rhWZ96rW-bH8rakQvFp2N8Fza0/view?usp=sharing)
I hope this will clarify the doubt. 

If you find this interesting, I can generate simulation pages for the SSD1306 examples and send a pull request to add links to the README.md file, so your users can easily find them with just one click.

This means that users can experiment with the library right within the browser, without having to download or install anything, and also verify their code before uploading it to the actual hardware.  

What are your thoughts?

lexus2k, 2020-08-11:
Hello,

The things you do are very cool and useful for many developers.
Of course I find that interesting. Thank you for clarifications.

Pull request is welcome

Puneethkumarsn, 2020-08-13:
Hi there,

This is to inform you that we are working on the links. We will share it shortly 🏇

Regards
Puneeth

Puneethkumarsn, 2020-08-16:
Hi There,

I am happy to share the direct simulation page for the library

https://wokwi.com/arduino/libraries/ssd1306/draw_bitmap

I will happy to create a PR and edit only the ReadMe file.

lexus2k, 2020-09-07:
PR is merged to master branch

closed 2020-09-11 by lexus2k