lexus2k/ssd1306 #110: ATMega328P compile error

coelner, 2020-08-08:
**Describe the bug**
```
/usr/bin/avr-ld: /tmp/arduino_build_795307/libraries/ssd1306/intf/vga/atmega328p/vga96x40.c.o (symbol from plugin): in function `ssd1306_vga_controller_96x40_init_no_output':
(.text+0x0): multiple definition of `s_vga_frames'; /tmp/arduino_build_795307/libraries/ssd1306/intf/vga/atmega328p/vga128x64.c.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

**To Reproduce**
Steps to reproduce the behavior:
1. open the ssd1306_demo.ino with the Artduino IDE
2. Try to compile it for an ATMega 328P board
3. Fail

**Expected behavior**
not to include the vga part

**Please complete the following information:**
 - 1.8.2
 - SSD1306
 - linux
 - Platform ATMega328P
 - IDE Arduino 1.8.13

**Additional context**
Add any other context about the problem here.

lexus2k, 2020-08-10:
Compilation is successful

```.txt
Sketch uses 9222 bytes (30%) of program storage space. Maximum is 30720 bytes.
Global variables use 424 bytes (20%) of dynamic memory, leaving 1624 bytes for local variables. Maximum is 2048 bytes.
```

I have Arduino IDE 1.8.10, the latest library sources from github https://github.com/lexus2k/ssd1306/tree/1.8_dev

coelner, 2020-08-10:
```
Linking everything together...
/usr/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_24440/ssd1306_demo.ino.elf /tmp/arduino_build_24440/sketch/sova.cpp.o /tmp/arduino_build_24440/sketch/ssd1306_demo.ino.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_16bit.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_1bit.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_8bit.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_fonts.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_generic.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_menu.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_uart.c.o /tmp/arduino_build_24440/libraries/ssd1306/nano_gfx.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/sprite_pool.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_console.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/intf/ssd1306_interface.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/i2c/ssd1306_i2c.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/i2c/ssd1306_i2c_embedded.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/i2c/ssd1306_i2c_twi.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/spi/ssd1306_spi.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/spi/ssd1306_spi_avr.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/spi/ssd1306_spi_usi.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/uart/ssd1306_uart_builtin.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/vga.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/atmega328p/vga128x64.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/atmega328p/vga96x40.c.o /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/esp32/CompositeOutput.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/esp32/vga128x64.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/composite_video.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/lcd_common.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/lcd_il9163.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/lcd_ili9341.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/lcd_pcd8544.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_sh1106.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_ssd1306.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_ssd1325.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_ssd1327.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_ssd1331.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_ssd1351.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/oled_template.c.o /tmp/arduino_build_24440/libraries/ssd1306/lcd/vga_monitor.c.o /tmp/arduino_build_24440/libraries/ssd1306/nano_engine/canvas.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/nano_engine/core.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/arduino/platform.cpp.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/avr/platform.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/esp/platform.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/linux/platform.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/mingw/platform.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/stm32/platform.c.o /tmp/arduino_build_24440/libraries/ssd1306/ssd1306_hal/template/platform.c.o /tmp/arduino_build_24440/libraries/Wire/Wire.cpp.o /tmp/arduino_build_24440/libraries/Wire/utility/twi.c.o /tmp/arduino_build_24440/libraries/SPI/SPI.cpp.o /tmp/arduino_build_24440/core/core.a -L/tmp/arduino_build_24440 -lm
/usr/bin/avr-ld: /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/atmega328p/vga96x40.c.o (symbol from plugin): in function `ssd1306_vga_controller_96x40_init_no_output':
(.text+0x0): multiple definition of `s_vga_frames'; /tmp/arduino_build_24440/libraries/ssd1306/intf/vga/atmega328p/vga128x64.c.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```
No, it does not compile, even for the 1.8._dev branch

```
diff --git a/src/intf/vga/atmega328p/vga128x64.c b/src/intf/vga/atmega328p/vga128x64.c
index 9cdfae0d..f177b212 100644
--- a/src/intf/vga/atmega328p/vga128x64.c
+++ b/src/intf/vga/atmega328p/vga128x64.c
@@ -28,7 +28,8 @@
 #include "lcd/lcd_common.h"
 #include "lcd/vga_commands.h"
 
-#if defined(CONFIG_VGA_AVAILABLE) && defined(CONFIG_VGA_ENABLE) && defined(__AVR_ATmega328P__)
+//#if defined(CONFIG_VGA_AVAILABLE) && defined(CONFIG_VGA_ENABLE) && defined(__AVR_ATmega328P__)
+#if !defined(CONFIG_VGA_AVAILABLE) || !defined(CONFIG_VGA_ENABLE) || !defined(__AVR_ATmega328P__)
 
 extern uint16_t ssd1306_color;
```
and now it compiles:
```
Linking everything together...
/usr/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_393627/ssd1306_demo.ino.elf /tmp/arduino_build_393627/sketch/sova.cpp.o /tmp/arduino_build_393627/sketch/ssd1306_demo.ino.cpp.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_16bit.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_1bit.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_8bit.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_fonts.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_generic.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_menu.c.o /tmp/arduino_build_393627/libraries/ssd1306/ssd1306_uart.c.o /tmp/arduino_build_393627/libraries/ssd1306/nano_gfx.cpp.o /tmp/arduino_build_393627/libraries/ssd1306/sprite_pool.cpp.o /tmp/arduino_build_393627 [... cut ...]
/usr/bin/avr-size -A /tmp/arduino_build_393627/ssd1306_demo.ino.elf
Sketch uses 9564 bytes (31%) of program storage space. Maximum is 30720 bytes.
Global variables use 430 bytes (20%) of dynamic memory, leaving 1618 bytes for local variables. Maximum is 2048 bytes.
```

I found the issue, it is the different avr-gcc version
```
.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc --version
ls (GNU coreutils) 8.32
/usr/bin/avr-gcc --version
avr-gcc (GCC) 10.2.0
```

I close the issue

closed 2020-08-10 by coelner

lexus2k, 2020-08-10:
How did you install so new version?

I reinstalled Arduino IDE, and it has only 7.3.0 avr-gcc version inside.

coelner, 2020-08-11:
https://wiki.archlinux.org/index.php/Arduino#AVR_Boards
Archlinux uses the upstream avr-gcc. I don't know why the `defined` statement have different results