lexus2k/ssd1306 #111: ATtiny standard SDA/SCL pins result in extreme slowdown

mstratman, 2020-08-13:
**Describe the bug**
When using PB2 and PB0 for SCL and SDA on an ATtiny85, with an  the display is updated extremely slowly.

** To reproduce: **

```
#define SSD1306_SCL 2 // physical pin 7
#define SSD1306_SDA 0 // physical pin 5
ssd1306_128x64_i2c_initEx(SSD1306_SCL, SSD1306_SDA, 0);
```

On the other hand, if you use PB3 and PB4 it works as expected.

** Screen: **
SSD1306, I2C, 128x64

mstratman, 2020-08-13:
My display's address was different than the hard-coded one.  If I set the address with a `#define SSD1306_SA 0x78` prior to including this library, it works as expected.

It's curious and strange to me that when using non-standard pins (e.g. PB3 and PB4) and NOT setting the correct address, everything works quickly and as expected. Perhaps something to do with emulating SPI via software rather than utilizing the hardware SPI? Just a barely-educated guess.

Anyhow, closing. Please reopen if that curiosity highlights an unexpected problem.

closed 2020-08-13 by mstratman

lexus2k, 2020-08-13:
Hello,

since you're using ssd1306 display over i2c, the use of ssd1306_128x64_i2c_initEx is correct.
0x3c is correct value, and your 0x78 is doubled 0x3c. So, you should use 0x3c in your sketch as i2c display address.
Software i2c algorithm for Attiny85 has no any differences for any pins, you use. It should work the same whatever pins you use (You can check that by looking into `ssd1306/src/intf/i2c/ssd1306_i2c_embedded.c`).

mstratman, 2020-08-13:
Thanks for looking at this.
After doing further research and experiments using your info, I was finding less consistent results, and realized the one thing I wasn't holding constant was the actual physical chips.

Turns out, despite selecting "internal 16MHz" in the Arduino IDE, it wasn't setting the fuses (looks like a boards.txt problem perhaps)... but if I manually set lfuse to 0xF1 it works as expected.

So all along it's simply been that the ATtiny's were running too slow.

Thanks again, and sorry for the false start.