lexus2k/ssd1306 #113: console.println() float problem

Phoenix1747, 2020-08-27:
**Describe the bug**
Just encountered this strange behaviour and it seems like the console can print "0.5" as expected but is unable to do so for "1/2". By trying to do so the screen only reads 0.

**To Reproduce**
This is the arduino sketch i tried this with. Uploaded on an ATTiny84a using an SSD1306 128x32 I2C display.
```cpp
#include <ssd1306.h>
#include <ssd1306_console.h>

Ssd1306Console console;

void setup() {
  /* Replace the line below with the display initialization function, you want to use */
  ssd1306_128x32_i2c_init();
  ssd1306_clearScreen();
  /* Set font to use with console */
  ssd1306_setFixedFont(ssd1306xled_font6x8);

  float a = 1/2;
  float b = 0.5;
  
  console.clear();
  console.println(a,2);
  console.println(b,2);
}

void loop() {
  //nothing
}
```

**Expected behavior**
Well... basically it should print 0.5 in both cases I presume.

**Please complete the following information:**
 - library version 1.8.2
 - SSD1306 128x32 I2C
 - Windows 10
 - ATTiny84a
 - Arduino IDE 1.8.13

lexus2k, 2020-08-27:
Hi there,

The issue relates basically to your programming language knowledge.
The library can do nothing with that.

Try this:
```.cpp
float a = 1f/2f;
```

closed 2020-08-27 by lexus2k

Phoenix1747, 2020-08-27:
Oops, thank you.