lexus2k/ssd1306 #117: lcd_common.c:100:17: error: Uninitialized variable: args [uninitvar]

Harvie, 2020-12-05:
Hello,
i've "accidentaly" checked the ssd1306 using cppcheck static analyzer and it seems quite ok, only one tiny error was found:

```
$ cppcheck --force . >/dev/null
src/lcd/lcd_common.c:100:17: error: Uninitialized variable: args [uninitvar]
            if (args < 0)
                ^
```

lexus2k, 2020-12-07:
Hi @Harvie 

Thank you for the finding, but my cppcheck doesn't show any warning.
I fixed this error (you can see commit in this issue).

Best regards

Harvie, 2020-12-07:
> Thank you for the finding, but my cppcheck doesn't show any warning.

i've just checked 1.8_dev branch and the warning has disappeared.

For the reference:

```
$ cppcheck --version
Cppcheck 2.2
```

lexus2k, 2020-12-15:
@Harvie 

Thank you very much.

commit 4ddcb632 (2020-12-07): Compiler warning fix #117
closed 2020-12-15 by lexus2k