lexus2k/ssd1306 #118: how to wire pcd8544

philippedc, 2020-12-14:
Hi Alexey, I have not yet used any Nokia5110 display and I would like to test it with your fabulous library. In fact I may upgrade some projects from ssd1306 to pcd8544.
I was looking to your demo example pcd8544_demo.ino and I have not understood where to declare and wire the display SCLK and DN (MOSI) pins for an Arduino Uno / Atmega328p....
Thanks

lexus2k, 2020-12-15:
Hello, Philippe,

Since you use Arduino Uno with Atmega328p, it doesn't allow to configure SPI pin numbers. Thus, you need to use only those allowed: 
[image]
MOSI: 11
SCLK: 13

Other pins as specified in demo code pcd8544_84x48_spi_init(3, 4, 5);  // FOR ATMEGA:
RST: 3
CE: 4 (instead of connecting to 4, you may pull down CE on display to GND)
DC: 5

philippedc, 2020-12-15:
Many thanks Alexey ! it works fine :)

philippedc, 2020-12-15:
in fact I was asking this question as the display first did not display anything.  I have ordered 5 of them, and unfortunately only one works perfectly. 
The strange thing - for me - is the comparison for the bad displays,  results
 between your library and carlosefr'one. There are days when electronics seem capricious and out of control....
[image]
[image]

lexus2k, 2020-12-15:
Hi,

I had the same issues with my Nokia displays. I see several points to look at:

**1. Power supply Voltage of lcd display.**

Since you have Arduino Uno, the mcu works at 16MHz. According to Atmega328p datasheet, mcu can work at 16MHz only if it's power supply voltage is 5V.
According to pcd8544 datasheet, allowable voltage for pcd8544 is 3.3V.
**I hope that your lcd display does that conversion internally**, but I'm not sure because I didn't see any level shifters on such display boards.

**2. Liquid crystals used**

pcd8544 is only display controller, it is only chip. And the display matrix itself can use different Liquid crystals, depending on manufacturer. And important parameter is threshold voltage of the crystals used in your display. pcd8544 controller has special command SetVOP to control that. So, you can try to change that value in the library to find out, which value fits better for you: [lcd_pcd8544.c](https://github.com/lexus2k/ssd1306/blob/master/src/lcd/lcd_pcd8544.c) (line 43):

```.cpp
    PCD8544_SETVOP | 0x16,  // Set vop contrast
    PCD8544_SETTEMP,
    PCD8544_SETBIAS | 0x04, // Set bias mode
```
Try to change 0x16 to other values, for example, check how the displays look with 0x26 and with 0x06.

My library uses BIAS 0x04 as recommended in PCD8544 datasheet, while carlosefr library uses BIAS 0x03. Next, VOP level for ssd1306 library is 0x16, while carlosefr library has 0x42. I agree, maybe VOP level is too low in ssd1306 library, and should be set higher.
And for some displays carlosefr library uses some undocumented mode 0x05. 

I would, say, if you have datasheet specific for your display, it would be very helpful.

Best regards

philippedc, 2020-12-17:
ok ! I will test. Many thanks

philippedc, 2021-01-10:
hi Aleksei, here are the results for some tests. In the meantime I received 10 other LCD, so the tests have been performed on a total of 15x "5110" lcd:
- 6 doesn't display anything, neither with carlosefr library nor yours.
- 5 are very well displaying with your library, I confirm the display and contrast is much better when VOP is increased: depending of the LCD the value can vary from 0x22 to 0x45 to get a good display with a good contrast. About the BIAS 0x03 in fine.
- now the surprise is that 4 displays are working with more or less contrast with carlosefr library, but do not show any sign of life with yours..... 
I'm trying to understand and compare the difference in both code, but I cannot even see the 0x42 and 0x03 values in the .cpp file... I can see 0x13 for BIAS line 87 and 0xc2 for VOP line 93.
Is it possible that these 4 displays use the ST7576 driver instead of the PCD8544 to explain this fail to display ?

by the way I'm very disappointed with this display. The best thing is that no backlight is needed to read it.... But it seems quite hard to work with. 
Thanks

lexus2k, 2021-01-10:
Hi again 👍 

> but I cannot even see the 0x42 and 0x03 values in the .cpp

Carlos library has line `this->send(PCD8544_CMD, 0xc2);  // default Vop (3.06 + 66 * 0.06 = 7V)`, where `0xc2 = 0x80 | 0x42`. 0x80 is command code (set vop), while 0x42 is command value.

The same thing for BIAS: `this->send(PCD8544_CMD, 0x13);  // bias system (1:48)`, where `0x13 = 0x10 | 0x03`. 0x10 is command code (set bias), while 0x03 is command value.

> Is it possible that these 4 displays use the ST7576 driver instead of the PCD8544 to explain this fail to display ?

Interesting, but I don't think that some displays in single batch have different controller. Maybe production quality is not stable?

> According to pcd8544 datasheet, allowable voltage for pcd8544 is 3.3V.

What about voltage levels for the display? Is VDD 3.3V? What about signal lines: are you using level shifters?

Do you have any datasheet from the manufacturer of these displays (or from supplier)?

> by the way I'm very disappointed with this display. The best thing is that no backlight is needed to read it....

Hmm, monochrome OLED displays do not use backlight also. But OLED displays have there own issue with pixels burning out. 
If you don't need dynamic picture, you could use E-Ink displays.

philippedc, 2021-01-11:
my projects for these ugly displays are ATmega328p based (8Mhz internal clock) powered with lithium-ion cells and a small solar panel. Even if 3.7-4.1V is yet higher than 3.3V, at the beginning displays were SSD1306 and they could support these over-voltages and still worked after a pair of years; replaced by PCD8544 since 27 days now they are still working....
As display results were not identical from a LCD display to another, I have tested them with the help of an Arduino Uno, powered the LCD under 3.3V but signal lines were under 5V. Now I'm making the tests with a serial 1K resistor for signal lines.
By the way I can assume this over-voltage could have burnt some displays, however I cannot explain why a part is working with Carlos library and not with yours.....
No I have not any datasheet from the manufacturer of these displays brought from aliexpress.
Yes E-Ink is a very good idea !

lexus2k, 2021-01-12:
> Even if 3.7-4.1V is yet higher than 3.3V, at the beginning displays were SSD1306 and they could support these over-voltages and still worked after a pair of years

Yeah, you're right 4.1V is not the problem. It should be OK.

> Now I'm making the tests with a serial 1K resistor for signal lines.

1K resistors cannot work correct as level shifters, but it still should be OK for the experiments.

> however I cannot explain why a part is working with Carlos library and not with yours

I can modify library with the values, used by Carlos library. Or you can do that by yourself. And if that doesn't help, I don't have ideas. Carlos library by default works as PCD8544.

philippedc, 2021-01-23:
Hi Aleksei, thank you for your help. I close the case, I have another issue so I will open a new one.

closed 2021-01-23 by philippedc