lexus2k/ssd1306 #119: Problem to the SDL simulator under windows

Darenn, 2021-01-11:
Ask any question, you have, regarding the library
Hi!

Thanks for this great library.
I'm under windows and trying to use the SDL simulator.

I've followed the [installation instructions](https://github.com/lexus2k/ssd1306/wiki/How-to-run-emulator-mode#compiling-example-and-running-emulation-in-windows), and I'm able to compile and run the example SDL project that is in the [tutorial ](https://lazyfoo.net/tutorials/SDL/01_hello_SDL/windows/mingw/index.php)linked. But I'm stuck with an error when trying to run the command `build_and_run.bat "games/lode_runner"`. 

Find attached the error I get [log.txt](https://github.com/lexus2k/ssd1306/files/5795979/log.txt).
Can you help me find what did I do wrong?

Thanks by advance!

lexus2k, 2021-01-12:
@Darenn 

Hi, I'm sorry for that, it seems that batch script is outdated, and maybe I need to create CMakeLists.txt file for the project.
Regarding the log file:

> gcc -std=c99 -I. -g -Os -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -Wno-error=deprecated-declarations   -c -o sdl_core.o sdl_core.c
> <builtin>: recipe for target 'sdl_core.o' failed

Do you have more logs? I see some generic error without any details. 
Did you install SDL2 libraries?

Best regards

Darenn, 2021-01-13:
@lexus2k thanks for your quick answer!

It's the error message I get when executing the command `build_and_run.bat "games/lode_runner" > log.txt`.
Do you know how I can get you more precise logs?

Yes I think I've installed the SDL2 correctly. As said in the tutorial you linked, I copied the content of the sld2 folder in one of my folder, and I'm able to correctly compile and run the example given **by specifying this folder on the compilation line in include** (folder is at : D:\Dev\mingw_lib\i686-w64-mingw32).

Maybe I missed a step?  it's true that I don't get how the batch/make can know where I installed the SDL2 without me specifying it. I don't see any more setup in the tutorial.

Thanks again!

lexus2k, 2021-01-15:
MINGW32 is so old.. and not supported.
BUT, can you check latest commit to 1.8_dev branch.
It should fix the issue

Darenn, 2021-01-15:
Oh sorry, am I using the wrong thing?
Yeah sure thanks, I will give it a go in the week end!

lexus2k, 2021-01-15:
If you installed it via Arduino Library Manager, then you're using correct version. But it has issue with SDL, you described.
I suggest you to get the code for the library from development branch 1.8_dev.

Darenn, 2021-01-15:
Ok I will download the lib and install it manually into arduino lib folder. Thanks!

Darenn, 2021-01-15:
Here is the new log I get with command `build_and_run.bat "games/lode_runner" > log.txt` using the 1.8 dev version.
[log.txt](https://github.com/lexus2k/ssd1306/files/5819632/log.txt)

lexus2k, 2021-01-18:
Thank you for the log. Will try to build using MINGW32 one more time on my laptop.

lexus2k, 2021-01-26:
@Darenn Unfortunately, I didn't find useful information in the log. And I cannot reproduce the issue on my PC.
If the problem is still actual for you, could you please add `CCFLAGS += -v` line to examples/Makefile.common file and provide more logs for me?

Darenn, 2021-01-27:
Hi!
Thanks for investigating, really seems like I installed something wrong if you can't reproduce it. Might be worth to try again the whole installation process.

I added the line CCFLAGS += -v at the end of examples/Makefile.common.

Here's the new log, not sure I did the right thing :

```
D:\T‚l‚chargements\ssd1306-master\ssd1306-master\tools>set project="games/lode_runner" 
        1 file(s) copied.

D:\T‚l‚chargements\ssd1306-master\ssd1306-master\tools>mingw32-make.exe -C ../examples -f Makefile.mingw32 EXTRA_CCFLAGS= SDL_EMULATION=y PROJECT="games/lode_runner" flash  
mingw32-make.exe: Entering directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/examples'
mingw32-make.exe -C ../src -f Makefile.mingw32 MCU= SDL_EMULATION=y \
                     ADAFRUIT=
mingw32-make.exe[1]: Entering directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/src'
mingw32-make.exe[1]: Nothing to be done for 'default'.
mingw32-make.exe[1]: Leaving directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/src'
mingw32-make.exe -C ../tools/sdl -f Makefile.mingw32 EXTRA_CPPFLAGS=""
mingw32-make.exe[1]: Entering directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/tools/sdl'
gcc -std=c99 -I. -g -Os -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -Wno-error=deprecated-declarations   -c -o sdl_core.o sdl_core.c
<builtin>: recipe for target 'sdl_core.o' failed
mingw32-make.exe[1]: Leaving directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/tools/sdl'
Makefile.linux:57: recipe for target 'ssd1306_sdl' failed
mingw32-make.exe: Leaving directory 'D:/Téléchargements/ssd1306-master/ssd1306-master/examples'
```

lexus2k, 2021-01-27:
I see some issue with sdl_core.c compilation. But my cppcheck tool doesn't show any warnings.
Which compiler version do you have in MINGW? Mine one has 4.8.3 version.

Darenn, 2021-01-27:
Not sure that's what you need but : 
`C:\Users\Darenn>g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=d:/dev/mingw/bin/../libexec/gcc/mingw32/6.3.0/lto-wrapper.exe
Target: mingw32
Configured with: ../src/gcc-6.3.0/configure --build=x86_64-pc-linux-gnu --host=mingw32 --with-gmp=/mingw --with-mpfr=/mingw --with-mpc=/mingw --with-isl=/mingw --prefix=/mingw --disable-win32-registry --target=mingw32 --with-arch=i586 --enable-languages=c,c++,objc,obj-c++,fortran,ada --with-pkgversion='MinGW.org GCC-6.3.0-1' --enable-static --enable-shared --enable-threads --with-dwarf2 --disable-sjlj-exceptions --enable-version-specific-runtime-libs --with-libiconv-prefix=/mingw --with-libintl-prefix=/mingw --enable-libstdcxx-debug --with-tune=generic --enable-libgomp --disable-libvtv --enable-nls
Thread model: win32
gcc version 6.3.0 (MinGW.org GCC-6.3.0-1)`

lexus2k, 2021-01-27:
@Darenn  What I'm trying to get is error code in compilation of sdl_core.o, since the log reports `recipe for target 'sdl_core.o' failed`. But actually the log contains only the fact of failure, but no failure description.
Mine MINGW is installed from http://win-builds.org/doku.php site.

Darenn, 2021-01-28:
Ok I understand. I don't know much about makefile, I tried added the -v everywhere I can find a compilation line but no changes.

It seems to execute this line to compile sdl_core.o
```
: gcc -std=c99 -I. -g -Os -Wall -Werror -ffunction-sections -fdata-sections -fno-exceptions -Wno-error=deprecated-declarations -v  -c -o sdl_core.o sdl_core.c
<builtin>: recipe for target 'sdl_core.o' failed
```

And the option is there but no message. I'm sorry I can't give you the info you need.
I will try reinstall when I have time.