lexus2k/ssd1306 #120: SSD1306 64*32

coelner, 2021-01-11:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
The same usage like the other screen sizes

**Describe alternatives you've considered**
There is a branch in the Adafruit SSD1306 library (https://github.com/mcauser/Adafruit_SSD1306/tree/esp8266-64x48)

**Additional context**
This is the display which I want to use: https://docs.wemos.cc/en/latest/d1_mini_shield/oled_0_66.html

lexus2k, 2021-01-14:
Hi @coelner 

Can you run ssd1306_demo and capture what it displays for me?

coelner, 2021-01-14:
Sorry, my v2.0.0 board went up in smoke. Unfortunately I have no replacement for this

lexus2k, 2021-01-14:
@coelner  Anyway, let me know, if you still need 64x48 or 64x32 display support.

coelner, 2021-01-15:
Yes, the 64x48 size would be nice.

lexus2k, 2021-02-04:
If it is suitable for you, ssd1306 is supported by [lcdgfx](https://github.com/lexus2k/lcdgfx) library. That is improved version of this library

closed 2021-02-04 by coelner