lexus2k/ssd1306 #122: 433MHz transmission trouble with pcd8544

philippedc, 2021-01-23:
Hi, I have several small projects running with Atmega328p and FS1000A / WL101 433MHz transmission : they detect a letter in box, the water level in tanks, etc....
Since now I used the OLED ssd1306 display for the receiver, but I thought better to replace them by Nokia 5110 just because the display is always on with a low power consumtion.
But I have a problem: by simply adapting the code from ssd1306 (driven by I2C) to pcd8544 ( driven by MOSI) makes the radio receiver losing the message from time to time. All messages are short, 7 bytes maxi. Sometime the message is received (and treated by the code) but sometime nothing happens.
For the radio transmission I use the very classical VirtualWire library (http://www.airspayce.com/mikem/arduino/VirtualWire).

About the receiver:
If some information is lost in reception, it may be because the process to update the display is much more longer for a pcd8544 than for a  ssd1306. Actual code makes the display completely updated every seconds, even for the ":" blink in the time display... Of course I will modify the code, so that only the place that need a change is updated. However, I would like some confirmation from you....
Can you confirm that the delay needed to update a display is much longer for a pcd8544 than for a  ssd1306 ? If yes approx how much time longer ?
If no, have you got an idea of the origin of the problem ?

About the transmitter:
I had the idea to add a pcd8544 with a radio transmitter. 
First, I had to take care in the order to declare in setup:

// initialise the IO and ISR of the radio transmitter
  vw_set_tx_pin(radioPin);     // transmitter initialization
  vw_setup(2000);              // transmission bits per seconds
  delay(1000);

// lcd initialization
  ssd1306_setFixedFont(ssd1306xled_font6x8);
  //pcd8544_84x48_spi_init(int8_t rstPin, int8_t scePin, int8_t dcPin);
  pcd8544_84x48_spi_init(3, 4, 5);
  ssd1306_clearScreen();
  ssd1306_printFixed (0, 0, "demarrage...", STYLE_NORMAL);
  delay(1000);
  ssd1306_clearScreen();

if I start by declaring the lcd first and then radio the lcd does not work (I don't know about the radio). But if I declare radio first and then the lcd, the lcd works but not the radio. 
Is it because both use serial ? How can I work around this ?
Thanks,

lexus2k, 2021-01-24:
Hi Philippe,

> Can you confirm that the delay needed to update a display is much longer for a pcd8544 than for a ssd1306 ? If yes approx how much time longer ?

I do not confirm that. pcd8544 works over SPI at default 8 MHz baud rate, while ssd1306 works over I2C at 400 kHz. So, pcd8544 interface should be almost 20 times faster.
I'm not familiar with VirtualWire library, but they say _`For testing purposes you can connect 2 VirtualWire instances directly, by connecting pin 12 of one to 11 of the other and vice versa, like this for a duplex connection`_ . So, I would advise you to check used pins in your project, because the same pins 11,12,13 on Atmega328p are usually used for SPI to communicate with lcd displays.

For example, you have pcd855 initialization like this: `pcd8544_84x48_spi_init(3, 4, 5);`, where 3, 4, 5 are RST, CS, D/C pins, but 11/13 pins are used implicitly by SPI bus.
RF module is initialized like `vw_set_tx_pin(radioPin)` in your code, radioPin value is unknown for me. I hope that you use some radioPin not related to SPI bus.

Next thing to check is timers/interrupts. Actually, ssd1306 library do not use any interrupts, and works via standard SPI interface. `cli/sei` are used only for i2c software implementation and that code is not used for pcd8544.

How does your loop cycle look like? Are there any delays in main program loop?

Best regards

philippedc, 2021-01-24:
`For testing purposes you can connect 2 VirtualWire instances directly, by connecting pin 12 of one to 11 of the other and vice versa, like this for a duplex connection . So, I would advise you to check used pins in your project, because the same pins 11,12,13 on Atmega328p are usually used for SPI to communicate with lcd displays.`

yes indeed I don't understand this testing purpose upon 2 wires as there is only 1 wire to go to the 433MHz transmitter, and also one wire coming from the receiver.... I made examples code with transmitter pin (radioPin)  #12 or #2
So it looks like a conflict in the SPI protocol.  I can confirm the problem keeps the same once I considerably optimized the display functions of the code. In fact in the manual the transmitter is pin #12 and receiver pin #11. I suppose that even if these pins are remapped the default pins stay unusable for something else :(  

Here is the code :
/*
433MHz receiver for the openned_mailbox_detector

from an idea of : https://www.carnetdumaker.net/articles/communiquer-sans-fil-en-433mhz-avec-la-bibliotheque-virtualwire-et-une-carte-arduino-genuino/

THE VirtualWire LIBRARY DOES NOT WORK WITH ATtiny85

Arduino Uno / ATmega328P wiring :
_________________________________

  A5 = PC5 in pin28 => SCL for SSD1306 OLED and DS1307 RTC + 24C32 EEPROM
  A4 = PC4 in pin27 => SDA for SSD1306 OLED and DS1307 RTC + 24C32 EEPROM
  12 = PB4 in pin18 => any traffic LED light
  10 = PB2 in pin16 => PCD8544 retro leds
   9 = PB1 in pin15 => DHT11 + 4.7Ko wired to Vcc
   8 = PB0 in pin14 => mailbox traffic buzzer
   7 = PD7 in pin13 => push button to set time and display change
   2 = PD2 in pin4  => data from 433MHz receiver

  13 = PB5 in pin19 => PCD8544 SCLK (default atmega328p SCK)
  11 = PB3 in pin17 => PCD8544 DN (default atmega328p MOSI)
   5 = PD5 in pin11 => PCD8544 D/C
   4 = PD4 in pin6  => PCD8544 SCE (option: may pull down SCE to GND)
   3 = PD3 in pin5  => PCD8544 RST

history 
 v5.0 - 15 dec 2020    - adaptation from ssd1306 oled to Nokia 5110 permanent display
 V5.1 - 17 jan 2021    - modify the change time process and display bug correction
 V6.0 - 23 jan 2021    - workarround by minimazing the display process
*/

// DS1307 addresses
const uint8_t DS1307_ADDRESS = 0x68;
const uint8_t DS1307_NVRAM_BASE = 0x08;
  

#include <LowPower.h>     // https://github.com/rocketscream/Low-Power
#include <VirtualWire.h>  // http://www.airspayce.com/mikem/arduino/VirtualWire/
#include "ssd1306.h"      // https://github.com/lexus2k/ssd1306 
#include <TimeLib.h>      // https://github.com/PaulStoffregen/Time
#include <DS1307RTC.h>    // https://github.com/PaulStoffregen/DS1307RTC
#include <DHT.h>          // https://github.com/adafruit/DHT-sensor-library
                          // https://github.com/adafruit/Adafruit_Sensor
#include <Wire.h>         // allow to direct communication with DS1307

#define RX_pin    2
#define pb_pin    7
#define buz_pin   8
#define dht_pin   9
#define lightPin 10
#define led_pin  12

// Create the DHT instance
DHT dht(dht_pin, DHT11);

int t0=0, memo_t0=0, h0;
int t1=0, memo_t1=0, h1;
int t2=0, memo_t2=0, h2;
int v0, v1, v2;
char heure[10];             // 1st line of display : the day of week + time
char th0[16];               // 2nd line of display : local temperature + hygrometry
char th1[16];               // 2nd line of display : outside temperature + hygrometry
char th2[16];               // alternatively the 2nd line of display : inside temperature + hygrometry
char line0[16], line1[16], line2[16];   // events display lines
[... 550 lines cut ...]
    ADMUX = _BV(REFS0) | _BV(MUX4) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
  #elif defined (__AVR_ATtiny24__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
    ADMUX = _BV(MUX5) | _BV(MUX0);
  #elif defined (__AVR_ATtiny25__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__)
    ADMUX = _BV(MUX3) | _BV(MUX2);
  #else
    ADMUX = _BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
  #endif  

  delay(2); // Wait for Vref to settle
  ADCSRA |= _BV(ADSC); // Start conversion
  while (bit_is_set(ADCSRA,ADSC)); // measuring

  uint8_t low  = ADCL; // must read ADCL first - it then locks ADCH  
  uint8_t high = ADCH; // unlocks both
  long result = (high<<8) | low;

  // autre écriture possible :
  //  result = ADCL;
  //  result |= ADCH<<8;

  //result = 1125300L / result; // Calculate Vcc (in mV); 1125300 = 1.1*1023*1000
  result = 11253L / result; // Calculate Vcc in dV (decivolt)
  return result; // Vcc in millivolts
}

lexus2k, 2021-01-27:
Hi

```.cpp
A5 = PC5 in pin28 => SCL for SSD1306 OLED and DS1307 RTC + 24C32 EEPROM
A4 = PC4 in pin27 => SDA for SSD1306 OLED and DS1307 RTC + 24C32 EEPROM
12 = PB4 in pin18 => any traffic LED light
10 = PB2 in pin16 => PCD8544 retro leds
9 = PB1 in pin15 => DHT11 + 4.7Ko wired to Vcc
8 = PB0 in pin14 => mailbox traffic buzzer
7 = PD7 in pin13 => push button to set time and display change
2 = PD2 in pin4 => data from 433MHz receiver

13 = PB5 in pin19 => PCD8544 SCLK (default atmega328p SCK)
11 = PB3 in pin17 => PCD8544 DN (default atmega328p MOSI)
5 = PD5 in pin11 => PCD8544 D/C
4 = PD4 in pin6 => PCD8544 SCE (option: may pull down SCE to GND)
3 = PD3 in pin5 => PCD8544 RST
```

Digital output 12 is still in use by standard SPI MISO. That can affect traffic LED. I don't think that there is conflict between ssd1306 library and Digital pin 2, used for 433MHz receiver.