lexus2k/ssd1306 #124: Avoid timer0 and Timer2 on nano

Shumatic, 2021-02-11:
Hi
Thanks for the lib. it is awsome
I am using an Nano with the lib and a 128x32 I2C.
afaik this lib uses othe Timer0 and Timer2, could this be avoided or is it I2C

lexus2k, 2021-02-11:
Hi,

Could you please provide code example that doesn't work for you?

Shumatic, 2021-02-14:
Hi,

Thank you for taking the time to answer.
I tried to strip my code, and I am not sure it is your code that is the problem, but I hope it is OK to send the code anyway, i understand if you dont answer.

I am using a Nano3 ATMEL 168
I found out if i reduced the buffer in my function subStr to 38  (#define MAX_STRING_LEN 38) i works. and the freeRam() report 135 bytes, when i increase MAX_STRING_LEN 39, it fails and the freeRam() report 151 bytes free RAM. I havent made this RAM function.

Any help would be appriciated, I like to know what i am missing.

`/**************************************************************************************************
  Filename:       ClickEncoder_timer2.ino

  Description:    ClickEncoder using Timer2

  Author:         Jesper Schumacher
  Last modified:  08-02-2021
*******Version history****************
  V1.0 08-02-2021 - Initial

**************************************************************************************************
*/
#define __FILENAME__ strrchr("\\" __FILE__, '\\') + 1

const byte numChars = 32; // for serial
char receivedChars[numChars];   // an array to store the received data from serial
boolean newData = false; // for serial
int DutyCycle = 50; // default
int cursorLCD = 9;

#include <ClickEncoder.h> //encoder
ClickEncoder *encoder; //encoder
long last; //encoder
int value; //encoder
int modeEncoder = 0; //encoder

//encoder interrupt
ISR(TIMER2_COMPA_vect) {
  encoder->service();
  //  digitalWrite(A0, !digitalRead(A0));
}

#include "ssd1306.h" // I2C A5 SCL - A4 SDA
int freeRam() {
  extern int __heap_start, *__brkval;
  int v;
  return (int)&v - (__brkval == 0 ? (int)&__heap_start : (int) __brkval);
}
void setup() {
  Serial.begin(115200);  Serial.print(F("\r\nFilename: ")); Serial.println(__FILENAME__);
  Serial.print(F(" Compiled:  "));  Serial.print(__DATE__);  Serial.print(" ");  Serial.println(__TIME__);

  pinMode(9, OUTPUT); //50% dutycycle
  pinMode(10, OUTPUT);

  //set timer2 interrupt at 1kHz
  TCCR2A = 0;// set entire TCCR2A register to 0
  TCCR2B = 0;// same for TCCR2B
  TCNT2  = 0;//initialize counter value to 0
  // set compare match register for 1khz increments
  OCR2A = 249;// = (16*10^6) / (8000*8) - 1 (must be <256)
  // turn on CTC mode
  TCCR2A |= (1 << WGM21);
  // Set CS21 bit for 8 prescaler
  TCCR2B = _BV(CS22);    //prescaler 32
  // enable timer compare interrupt
  TIMSK2 |= (1 << OCIE2A);
  sei();//allow interrupts

  encoder = new ClickEncoder(A1, A0, 6, 4); //encoder
[... 169 lines cut ...]
   @brief   split string on given delimiter

   @param   str     input string to splitted
            delim   char to use as delimiter
            index   block to return,1 is first block BEFORE delim.

   @return  Block between chosen delimiters
   /**********************************************************************************************
*/
char* subStr (char* str, const char *delim, int index) {
#define MAX_STRING_LEN 100
  char *act, *sub, *ptr;
  static char copy[MAX_STRING_LEN];
  int i;
  // Since strtok consumes the first arg, make a copy
  strcpy(copy, str); 
  copy[MAX_STRING_LEN] = '\0';
  for (i = 1, act = copy; i <= index; i++, act = NULL) {
    ////Serial.print(".");
    sub = strtok_r(act, delim, &ptr);
    if (sub == NULL) break;
  }
  return sub;
} //subStr
`

lexus2k, 2021-02-15:
Hi,

Unfortunately, I cannot compile the example. The system doesn't know anything about `#include <ClickEncoder.h>`.
As for ssd1306 library, the only place, where interrupt handlers are declared for VGA mode is `vga_isr.h` header, which is not included by library cpp/c files. The only place, where this header is included, is examples. So, I don't think that library affects TIMER0 in your case.

As for free ram. What is reported by Arduino, when it compiles your code?

Shumatic, 2021-02-15:
Hi

No, it is an external lib https://github.com/0xPIT/encoder.
i Like it becurse it handle both button and encoder (with acceleration)
Arduino dosnt report freeRam, only flash, and the sketchuse app. 10500byte
/j

lexus2k, 2021-02-15:
> Arduino dosnt report freeRam, only flash, and the sketchuse app. 10500byte

That's very strange. My Arduino IDE 1.8.3 for Arduino Nano based on Atmega328p shows:

```.txt
Sketch uses 5826 bytes (18%) of program storage space. Maximum is 32256 bytes.
Global variables use 360 bytes (17%) of dynamic memory, leaving 1688 bytes for local variables. Maximum is 2048 bytes.
```

So, it actually displays how much memory is required for global variables, the rest 1688 can be used for local variable and function calls (all of these ones are allocated in Stack)

Shumatic, 2021-02-18:
Hi Aleksei
Yes i had comprehensive output on, so i didnt notice. then i wouldn't have posted this in the first place, as it clearly say
Sketch uses 10572 bytes (73%) of program storage space. Maximum is 14336 bytes.
Global variables use 862 bytes (84%) of dynamic memory, leaving 162 bytes for local variables. Maximum is 1024 bytes.
**Low memory available, stability problems may occur.**
Theres a little diff, this says 162 b, FreeRAM states 151 b

Thanks for you positive attitude and explanation.
Sorry for wasting you timeon such triviallity. 
BR Jesper

lexus2k, 2021-02-18:
Hi

You're welcome. Did you solve the original problem with timers?

Best regards

Shumatic, 2021-02-19:
Hi
Yes, it wasn't a timer og SSD issue, it was only a too big program for a too small Proc. :-)
Normally ClickEncoder use a lib called TimerOne, that ...... use timer1
The code i posted here, use Timer2, i have also teste with Timer0, but that them millis() are gone.

lexus2k, 2021-02-20:
Thank you for the great explanation and a lot of code examples.
I think all those information will be very helpful for everyone.

For now, I'm closing the issue. Feel free to open the new one if you have any problems.

Best regards

closed 2021-02-20 by lexus2k