lexus2k/ssd1306 #127: Library directory issue in PlatformIO.

adi-maxx, 2021-03-31:
Good day to you,

I've stumbled upon an error when I try to compile an example of this library, sh1106_demo to be specific. The error stated that it did not found the directory for the "Wire.h" even though it exist in the library directory.
I wonder why this happen?

Additional information--
library version 1.8.0

method of using the library in PlatformIO:
by pasting the library in the lib directory and add library path in the configuration.

------------

[image]

[Terminal Log.txt](https://github.com/lexus2k/ssd1306/files/6233299/Terminal.Log.txt)

Regards,

lexus2k, 2021-04-02:
Hi,

Unfortunately, I don't use platform io directly. I believe it dependency issue. 
You can find similar issues and solutions over internet: https://community.platformio.org/t/wire-h-not-found-platformio-latest-version/2046
If you solve the problem, please make a note here about the solution.

Best regards

adi-maxx, 2021-04-04:
Oh I see, fortunately I did manage to solve this issue for my project by adding #include <Wire.h> and <SPI.h>. Which make use the header file from the platformIO directory instead. By adding the #include <Wire.h> and <SPI.h>, ssd1306 library works fine.

Cheers,

closed 2021-04-04 by adi-maxx