lexus2k/ssd1306 #128: How can I use the library with esp-idf?

alexmaron81, 2021-04-27:
I would like to control the ssd1351 via SPI.
What would the main.c look like for this?

lexus2k, 2021-04-27:
Hi,

As pointed on the main page in the [section "Setting up"](https://github.com/lexus2k/ssd1306#setting-up):

```
For esp32:

    * Download source from https://github.com/lexus2k/ssd1306
    * Put downloaded sources to components/ssd1306/ folder.
    * Compile your project as described in ESP-IDF build system documentation
```

alexmaron81, 2021-04-29:
That's what I did too.
But I have now found out that the GPIO 0 did not work for the CS pin in the ESP-IDF.
I took a different pin for it.
It's running now

lexus2k, 2021-04-29:
Oh, that's the bug. I will try to fix it. Probably, I know the root cause.

chandanchan0086, 2021-05-07:
hi @alexmaron81 can u please send me example code for how to use this library using esp-idf?