lexus2k/ssd1306 #135: Can this be used on Linux?

sgjava, 2021-12-01:
Specifically Armbian?? Do you need to provide your own I2C and SPI implementations?

lexus2k, 2021-12-04:
Hi Steven,

Yes, you can use the library in linux. Current implementation supports i2cdev and spidev modules.
From the main page:
[image]

sgjava, 2021-12-04:
So I just run make from project root? GCC?? This would be on Armbian and a NanoPI Duo (ARM32 5.x kernel).

lexus2k, 2021-12-05:
Yes, you can run make as:

`make -C ../src -f Makefile.linux`

sgjava, 2021-12-05:
OK, so I ran that from tools and it looks like object files are in build. Does it create any executables?

lexus2k, 2021-12-05:
Maybe, you will find useful information on this page: https://github.com/lexus2k/ssd1306/wiki/How-to-run-emulator-mode

> Does it create any executables?

Yes, it does. Please check `./bld` directory

sgjava, 2021-12-09:
OK, simulator up and running, thanks.

closed 2021-12-09 by sgjava

lexus2k, 2021-12-09:
Thank you for the feedback 
However, I agree that the sysfs API is outdated and I need to switch to newer API in Linux.

sgjava, 2021-12-09:
So I'm want to wrap your library in JNI for Java similar to U8g2 https://github.com/sgjava/javauio/tree/main/u8g2, but I'm trying to get it setup in Eclipse (just to compile the C stuff). What IDE are you using for dev? Do you have instructions for devs? I already switched U8gw arm-linux to the userspace APIs https://github.com/olikraus/u8g2/tree/master/sys/arm-linux

lexus2k, 2021-12-10:
I do not use any IDE for this project. However it has `Makefile` and `CMakeLists.txt` files. And I think those files should be enough for IDE like Eclipse (https://mcuoneclipse.com/2017/07/22/tutorial-makefile-projects-with-eclipse/, https://stackoverflow.com/questions/9453851/how-to-configure-eclipse-cdt-for-cmake)

sgjava, 2021-12-10:
I figured you had your own build system :) I'll look at Makefile. By default Eclipse tried to run "all" target and it cannot find that in root Makefile. I build the arm-linux port of U8g2 in Eclipse C project. Let me poke around.

sgjava, 2021-12-10:
OK got it, had to pass -C src -f Makefile.linux in the args, but now I can build in Eclipse. Next I need to figure out how to change pins for my SBC.

sgjava, 2021-12-10:
OK, I see where you are using __KERNEL__ for the IO stuff. platform.c should be pretty easy to convert. Let me test as is first once I figure out how to run hello world on my text board. For now I can compile and edit under x86 Ubuntu.

sgjava, 2021-12-10:
OK, got ./build_and_run.sh -p linux -f demos/ssd1306_demo to run on Pine A64 with I2C display. Guess I didn't have to define pins since it's using hardware I2C. One thing nice about U8g2 is you can drive I2C and SPI using software (bit banging GPIO pins). I just need to figure out how to get this all building in Eclipse.