lexus2k/ssd1306 #136: SSD1331 demo failing

sgjava, 2021-12-11:
`sudo ./build_and_run.sh -p linux -f demos/ssd1331_demo 
make: Entering directory '/home/servadmin/ssd1306/examples'
make -C ../src -f Makefile.linux MCU= SDL_EMULATION= \
                     ADAFRUIT=
make[1]: Entering directory '/home/servadmin/ssd1306/src'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/home/servadmin/ssd1306/src'
mkdir -p ../bld/demos/
cc -o ../bld/demos/ssd1331_demo.out -g -Os -w -ffreestanding -MD -g -Os -w -ffreestanding -I./demos/ssd1331_demo -I../src -Wall -Werror -Wl,--gc-sections -ffunction-sections -fdata-sections  ../bld/main.o ../bld/demos/ssd1331_demo/sova.o ../bld/demos/ssd1331_demo/ssd1331_demo.o -L../bld -lssd1306
../bld/demos/ssd1331_demo.out
Failed to initialize SPI: No such file or directory!
^Cmake: *** [Makefile.linux:52: flash] Interrupt
`
I assume there's a way to set SPI device and pis for DC and RESET? Also I shouldn't have to run as root as I have the permissions set for non -root user on SPI device.

lexus2k, 2022-01-06:
Do you have spidev kernel module loaded?
Does the application have access privileges to kernel module driver?
What is the configuration of the target system?

lexus2k, 2022-03-10:
Yes, it use spidev from userspace, but accessing to spi char devices requires root privileges

sgjava, 2022-03-10:
I just use a script to set permissions https://github.com/sgjava/javauio/blob/main/scripts/uio-permissions.sh