lexus2k/ssd1306 #137: How to use Software I2C with this library?

papsukkal09, 2022-02-21:
Please provide the following information:
## library version 1.8.2
## LCD display type ssd1306
## Steps to reproduce the issue
## Expected behavior
## Actual behavior

How can I define software I2C with this library on arduino uno (atmega328p).

#define CONFIG_SOFTWARE_I2C_AVAILABLE
#define CONFIG_SOFTWARE_I2C_ENABLE

I used this defination but it does nothing.

lexus2k, 2022-03-10:
Hi, 

you need manually initialize software i2c via `void ssd1306_i2cInit_Embedded(int8_t scl, int8_t sda, uint8_t sa);`

ishyesh, 2023-10-13:
First, insert 
`#include "intf/i2c/ssd1306_i2c.h` into `ssd1305.h`.

Also, looking at the source, for Arduino Uno 328p, you have to use ports on PORTC.  (`ssd1306_i2c_embedded.c`).