lexus2k/ssd1306 #138: Invalid function call in CompositeOutput.cpp (doesn't compile for ESP32)

jelli123, 2022-02-25:
It seems that the argument list when calling an deprecated Espressif 32 (PlatformIO) function has been replaced by that of the new function, but the function name has not been corrected.

/src/intf/vga/esp32/CompositeOutput.cpp

 - library version 1.8.3
 - Platform Esp32 (Espressif 32: development platform for PlatformIO)

Suggestion:
in
`void CompositeOutput::check_buffer()`
replace
`i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);`
with
`i2s_write(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);`

lexus2k, 2022-03-01:
Hi,
can you please download latest code from the master branch.
It already has the fix [78c2dcd](https://github.com/lexus2k/ssd1306/commit/78c2dcd58ff3e6783360437410327dbd391e9599)

jelli123, 2022-03-02:
sorry, I found out that this is the same as issue #132 and it was fixed last October in the GitHub master branch.

BUT, when I load version 1.8.3 in PlatformIO, I get a different /src/intf/vga/esp32/CompositeOutput.cpp (line 275).

> void CompositeOutput::check_buffer()
> {
> if (m_ptr == m_end)
> {
> size_t bytes_written;
> i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);
> m_ptr = line;
> }
> }

There it shows ` i2s_write_bytes` and in the GitHub master it shows ` i2s_write`.
Strange...

lexus2k, 2022-03-04:
That's very strange since the fix is included to 1.8.3 according to the github:

[image]

lexus2k, 2025-05-20:
Closed as not reproducible

closed 2025-05-20 by lexus2k