lexus2k/ssd1306 #141: Working on a IP Camera - Anyka

ricardojlrufino, 2022-04-03:
Just passing by to let you know that I managed to use your implementation on an IP camera lol

Linux 3.4.35 #5 Wed Sep 18 15:36:19 CST 2019 armv5tejl GNU/Linux

[image]

lexus2k, 2022-04-03:
Hi,

Thank you. That's amazing. I'm glad if my library helped you.
On the picture I see that the display is connected over i2c bus, which gives something around 300k throughput for payload data in 400kHz mode. Did you try spi displays? They are much faster (sometimes up to 30Mbps).
What are you using display for?

ricardojlrufino, 2022-04-04:
HI,

This display I'm using doesn't have RGB, so I couldn't test more advanced things.
This was just an I2C integration test with a generic IP camera... 

This integration I implemented using i2C Software with a relatively simple implementation, I would have to implement something similar for SPI... not yet tried.

https://ricardojlrufino.wordpress.com/2022/04/03/hack-ipcam-anyka-oled-display-on-cam/

https://ricardojlrufino.wordpress.com/2022/04/01/hack-ipcam-anyka-compiling-i2c-software-kernel-driver/