lexus2k/ssd1306 #142: pgmspace.h issue since 1.8.3

Polohpi, 2022-04-07:
**Describe the bug**
used in a sketch compiled for cortex M0 chips. Cant compile code. Works with 1.8.2.

**Expected behavior**
Code cant compile for AMD cortex m0.

**Please complete the following information:**
 - library version 1.8.3
 - LCD display type : oled 128 64
 - Debian
 - seeeduino xiao 
 - arduino ide

**Additional context**
libraries/ssd1306/src/ssd1306_hal/arduino/io.h:45:14: fatal error: api/deprecated-avr-comp/avr/pgmspace.h: No such file or directory
     #include <api/deprecated-avr-comp/avr/pgmspace.h>
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Erreur de compilation pour la carte Seeeduino XIAO

lexus2k, 2022-04-08:
By the way, what's is the version of board support package for SAMD controllers?

Polohpi, 2022-04-08:
> By the way, what's is the version of board support package for SAMD controllers?

1.8.2

Polohpi, 2022-04-08:
> Hi,
> 
> please try to comment out the lines in [src/ssd1306_hal/arduino/io.h](https://github.com/lexus2k/ssd1306/blob/master/src/ssd1306_hal/arduino/io.h) source code:
> 
> ```c++
>     #include <avr/pgmspace.h>
>     #include <avr/interrupt.h>
>     //#elif defined(ARDUINO_ARCH_SAMD)
>     //#include <api/deprecated-avr-comp/avr/pgmspace.h>
>     //#include <api/deprecated-avr-comp/avr/interrupt.h>
>     #else
>     #include <avr/pgmspace.h>
>     #include <avr/interrupt.h>
>     #endif
> ```
> 
> Let me know what it gives you.

It did work ! thanks. What was the problem with those lines ?

lexus2k, 2022-04-09:
Board support package version is not the same thing as Arduino IDE version. Arduino IDE has built-in manager as you know, where you can install support packages for different hardware. Those packages are supported by numerous teams. 

Since a developers  work on different versions of board support packages, some source files for the same SAMD board may have different location. Probably, you use some old version of the board support packages, because the file `<avr/pgmspace.h>` was moved to new location in newer releases  `<api/deprecated-avr-comp/avr/pgmspace.h>`.