lexus2k/ssd1306 #147: Add support for recent series of attiny series.

JasXSL, 2022-06-01:
**Is your feature request related to a problem? Please describe.**
Thanks to [megatiny](https://github.com/SpenceKonde/megaTinyCore) you can use the newer series of attiny 1604, 3227 etc with arduino.

**Describe the solution you'd like**
Support for this library isn't too hard to add in. I forked the repo and added the following to io.h
```
#elif defined(__AVR_ATtinyxy4__) || defined(__AVR_ATtinyxy2__) || defined(__AVR_ATtinyxy6__) || defined(__AVR_ATtinyxy7__)
    /** The macro is defined when i2c Wire library is available */
    #define CONFIG_PLATFORM_I2C_AVAILABLE
    /** The macro is defined when SPI library is available */
    #define CONFIG_PLATFORM_SPI_AVAILABLE
    /** Define lcdint as smallest types to reduce memo usage on tiny controllers. *
     * Remember, that this can cause issues with large lcd displays, i.e. 320x240*/
    #define LCDINT_TYPES_DEFINED
    /** This is for Attiny controllers */
    typedef int8_t lcdint_t;
    /** This is for Attiny controllers */
    typedef uint8_t lcduint_t;
```

Nothing else was needed to run a simple 0.93" OLED. **However, there are probably some things to think about before any eventual pull request:**

* These microcontrollers come with a range between 2k and 32k of flash. Deciding where to use uint8_t or not may need consideration.
* **SSD1306_WIRE_CLOCK_CONFIGURABLE** may work. But setting the clock that high did cause other i2c devices to stop working, so I left it out.

tpimh, 2022-06-11:
Thank you for your patch, it should be very useful.

Designing a circuit with ATtiny826 now, I'm wondering about the feasibility of using this library with it. My display is SSD1306 SPI 128x32. I guess, half RAM would be sacrifised for the display buffer (no double buffering).

I wonder what displays can be used with the 2k RAM microcontrollers.

lexus2k, 2022-10-12:
@JasXSL Hi

Can  you create pull request for you change?