lexus2k/ssd1306 #148: Support Pi Pico / RP2040 Microcontrollers

alijani1, 2022-06-26:
Does not seem the library support the RP2040 controllers like the Pi Pico.

lexus2k, 2022-10-12:
Hi
If you let me know some more details on what IDE you use for Pi Pico, what BSP is used, that would be very helpful

RockaBen99, 2022-12-11:
I have also been trying to use this library with a Pico, using Mbed OS (tried with both version 2.5.2 and version 3.4.1). When attempting to upload through Arduino IDE 1.8.13 on Debian (specifically Raspbian) Linux, I get this error:

> arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/benedict/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/benedict/.arduino15/packages -libraries /home/benedict/Arduino/libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=2E8A_00C0 -ide-version=10813 -build-path /tmp/arduino_build_482146 -warnings=none -build-cache /tmp/arduino_cache_934760 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/benedict/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=/home/benedict/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.bossac.path=/home/benedict/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=/home/benedict/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.rp2040tools.path=/home/benedict/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=/home/benedict/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.dfu-util.path=/home/benedict/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=/home/benedict/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -verbose /tmp/arduino_modified_sketch_181006/ssd1306_demo.ino
arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/benedict/.arduino15/packages -tools /usr/share/arduino/hardware/tools/avr -tools /home/benedict/.arduino15/packages -libraries /home/benedict/Arduino/libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=2E8A_00C0 -ide-version=10813 -build-path /tmp/arduino_build_482146 -warnings=none -build-cache /tmp/arduino_cache_934760 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs=runtime.tools.openocd.path=/home/benedict/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.openocd-0.11.0-arduino2.path=/home/benedict/.arduino15/packages/arduino/tools/openocd/0.11.0-arduino2 -prefs=runtime.tools.bossac.path=/home/benedict/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=/home/benedict/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino2 -prefs=runtime.tools.rp2040tools.path=/home/benedict/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=/home/benedict/.arduino15/packages/arduino/tools/rp2040tools/1.0.2 -prefs=runtime.tools.dfu-util.path=/home/benedict/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=/home/benedict/.arduino15/packages/arduino/tools/dfu-util/0.10.0-arduino1 -verbose /tmp/arduino_modified_sketch_181006/ssd1306_demo.ino
Using board 'pico' from platform in folder: /home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2
Using core 'arduino' from platform in folder: /home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2
Build options changed, rebuilding all
Detecting libraries used...
"/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO/defines.txt" "@/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO/cxxflags.txt" -DARDUINO_ARCH_RP2040  -mcpu=cortex-m0plus   -w -x c++ -E -CC -DARDUINO=10813 -DARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_MBED_RP2040 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-I/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/cores/arduino" "-I/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO"    "-I/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/cores/arduino/api/deprecated" "-I/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/cores/arduino/api/deprecated-avr-comp" "-iprefix/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/cores/arduino" "@/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO/includes.txt" "/tmp/arduino_build_482146/sketch/ssd1306_demo.ino.cpp" -o "/dev/null"
"/home/benedict/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO/defines.txt" "@/home/benedict/.arduino15/packages/arduino/hardware/mbed_rp2040/2.5.2/variants/RASPBERRY_PI_PICO/cxxflags.txt" -DARDUINO_ARCH_RP2040  -mcpu=cortex-m0plus   -w -x c++ -E -CC -DAR [... cut ...]
/home/benedict/Arduino/libraries/ssd1306/src/intf/i2c/ssd1306_i2c_embedded.c:33:10: fatal error: util/delay_basic.h: No such file or directory
 #include <util/delay_basic.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
Using library ssd1306 at version 1.8.3 in folder: /home/benedict/Arduino/libraries/ssd1306 
exit status 1
Error compiling for board Raspberry Pi Pico.

A similar error happened to someone attempting to use the SoftwareSerial library with the Arduino Nano RP2040 Connect: https://forum.arduino.cc/t/softwareserial-file-util-delay-basic-h-not-found/897597

alijani1, 2023-05-13:
> Hi If you let me know some more details on what IDE you use for Pi Pico, what BSP is used, that would be very helpful

Hi.  I prefer PlatformIO with VS Code but Arduino IDE would also be fine.