lexus2k/ssd1306 #152: I2C definition for ESP32 Espressif

kuzmanzs, 2022-09-11:
Hi,

I would like to use your display library. I have ESP32 expressif project with a make file.

I got following warning (and dispaly does not show anything)

warning: the default selection I2S (undefined) of <choice> (defined at main/Kconfig.projbuild:23) is not contained in the choice

It tried to define silly in my "Kconfig.projbuild" but it does not work:
menuconfig I2C
    string "SDA"
               default "21"
     string "SDL"
               default "22"
endmenu

What would the proper way to define it?

lexus2k, 2022-09-12:
@kuzmanzs 

hello. I2S or I2C ? What is the exact error message?

kuzmanzs, 2022-09-12:
it is I2C with 0.96" Oled display.

I have only this warning:
"warning: the default selection I2S (undefined) of (defined at main/Kconfig.projbuild:23) is not contained in the choice"

Actually the firmware builds, but display does not show anything. I have put some text on it from the examples.
I have connected my display to default SDA(21)/SCL(22) /ESP32 Expressif/ and power. With Arduino framework it works well.

Hmm, yes you are right it is I2S warning and not I2C.. 
I am not big expert at ESP IDF area, maybe should I "enable" I2C in menuconfig?

I have sdkconfig and Kconfig..

kuzmanzs, 2022-09-12:
thx for the lib

closed 2022-09-12 by kuzmanzs