lexus2k/ssd1306 #155: Potential Conflict AHTX10/AHT10 and LCDGFX

schwos, 2023-08-01:
First, not 100% sure if this is a bug or just over looking a setting/configuration within the code, but been looking up/down and sideways to figure out what is causing the conflict between LCDGFX and AHT10 with respects to the i2C bus. 

Code is running on a Arduino with a SSD1306 LCD Screen, two INA260, AHT10, and Infrared Sensor. Everything works fine except the AHT10. We have attempt the AHT10 library as well AHTX0 library both fail to operate. 

Don't know if anyone has run into a similar issue and found a work around or not. We moved to the LCDGFX library because of its size. When using the SSD1306 and Adafruit_GFX libraries with all the other libraries paired down we would run out have available ram.

Quick summary of code which the AHT10 will not connect. If I change the order it will connect but post display.begin(); it can't read its data.. 

`#include "lcdgfx.h"
#include "lcdgfx_gui.h"
#include <AHT10.h>
AHT10 myAHT10(AHT10_ADDRESS_0x38);
uint8_t readStatus=0;

void setup()
{
Wire.begin();
display.begin();

while (myAHT10.begin() != true)
    {
        //displayErrors("AHT10 failed..");  //display error on LCD 
        Serial.println(F("AHT10 not connected or fail to load calibration coefficient"));
        delay(5000);
    }
}

loop{
}` 

Appreciate any suggestions or guidance as I continue to read around..

schwos, 2023-08-02:
After much testing, scope reading we isolated the issue to the INA260 source code vs LCDGFX. So this at the moment can be closed and not a LCDGFX potential bug.

closed 2023-08-02 by schwos

lexus2k, 2023-08-02:
@schwos  Anyway, let me know if you need some help and change to the library
As for the INA260 and ssd1306 display they should not have any intersection on i2c addresses. ssd1306 library holds i2c line only if you send some command to the display.