lexus2k/ssd1306 #159: Error compiling ssd1306_demo

sidtupper, 2023-11-29:
**Describe the bug**
Compile error in ssd1306_demo

E:\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp: In member function 'void CompositeOutput::check_buffer()':
E:\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp:280:9: error: 'i2s_write_bytes' was not declared in this scope
         i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);
         ^~~~~~~~~~~~~~~
E:\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp:280:9: note: suggested alternative: 'i2s_write_expand'
         i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);
         ^~~~~~~~~~~~~~~
         i2s_write_expand

exit status 1

Compilation error: exit status 1

**To Reproduce**
Steps to reproduce the behavior:
- In Arduino IDA / Manage Libraries install ssd1306 
- load Examples/ssd1306/demos/ssd1306_demo
- Verify/Compile

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Please complete the following information:**
1.8.3 - library version
ssd1306 - LCD display type
Windows 11 - OS [e.g. linux, windows]
ESP32 - Platform [e.g. Atmega328p, esp32, etc.]
Arduino IDE 2.2.1 - IDE if using some

**Additional context**
Add any other context about the problem here.

anonymous721, 2023-12-29:
This isn't just with the demo. Even an empty default sketch plus just the initial `#include "ssd1306.h"` gives me the same error.
Same information (package v1.8.3, Arduino IDE v2.2.1, and an ESP32 device) except Linux instead of Windows.

M1rk0, 2024-01-15:
Having the same error with ESP Wroom 32 Dev Board, Arduino IDE 2.2.1, Windows 11.

lexus2k, 2024-03-09:
I can confirm that I observe the same issues on the latest releases:

```.txt
c:\Users\alexe\OneDrive\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp: In member function 'void CompositeOutput::check_buffer()':
c:\Users\alexe\OneDrive\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp:280:9: error: 'i2s_write_bytes' was not declared in this scope
         i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);
         ^~~~~~~~~~~~~~~
c:\Users\alexe\OneDrive\Documents\Arduino\libraries\ssd1306\src\intf\vga\esp32\CompositeOutput.cpp:280:9: note: suggested alternative: 'i2s_write_expand'
         i2s_write_bytes(I2S_PORT, (char*)line, sizeof(uint16_t) * (m_end - line), &bytes_written, portMAX_DELAY);
         ^~~~~~~~~~~~~~~
         i2s_write_expand
```

It will be fixed

lexus2k, 2025-05-20:
@M1rk0 @sidtupper Is the issue still reproducible with master branch?