lexus2k/ssd1306 #162: None of examples load on attiny44

jtannenba, 2024-03-08:
Ask any question, you have, regarding the library
All the examples I've tried say the program is too big for the chip.

I'm using arduino ver 1.8.19
ver 1.8.3 of the SSD1306 library  and
ver 1.0.2 of David A Mellis attiny

Should it load?

lexus2k, 2024-03-09:
Hello @jtannenba 

I have re-checked compilation, and I can confirm that the ssd1306_demo project can be built for AttnyX4 series.
The memory requirements depend on the mcu you are using. Some examples can fit Attiny84 mcu only.

Best regards,
Alexey

jtannenba, 2024-03-09:
ok, I was trying a attiny44.

closed 2024-03-09 by jtannenba

lexus2k, 2024-03-09:
You can remove some demo parts in the examples.
If you need any help with attiny44, let me know