lexus2k/ssd1306 #171: menu selection

mannitb303, 2025-07-19:
Hello, i am Testing your library gor my Midi Controller. I use your menu function and it works great but how i can get what Menü Item is selectect? 
Thank you.

lexus2k, 2025-07-21:
Can you please provide screenshot and pictures?
Did you try to run menu_demo example?

mannitb303, 2025-07-22:
Hi, thank you for your reply. I found the answer in the ssd1306_Demo. Yes i tryed the menu_demo and there is a line that i can request the current position of menu cursor by calling ssd1306_menuSelection() but i dont know how i can work with that.  But in the ssd1306_demo i found it how to work with the menuSelection it is done with cases. 
Now it works.
Thank you for the great Library.